runmyrobot/letsrobot_ios #1: Ideas List

Sherlouk, 2017-06-03:
- Favourite Bots
  - Allow users to favourite bots, they will show up first in the chooser when live
- Search Bots
  - As more bots get added, a search feature will be handy to quickly filter through the list
- Filter Chat for "This Bot Only" vs "All Bots"
  - This will allow users to quickly toggle between seeing all the chat from everybody, or just people in the same room/robot as you.
- Client side profanity filter
  - The web handled all profanity filters on the client side, we should add an equivelent system for the iOS client
  - Should also add the ability to turn it on/off as a user
- Create a splash screen and pre-loader
  - Currently you just get an empty screen whilst the config and socket are loaded and connected. We should make this experience a bit tidier
- Update the robot chooser to better distinguish between online/offline bots. Potentially defaulting to only show online bots (as you can't do anything with offline ones!)
- Update the chat area to better use the space and have the main message inline and wrap around the author rather.
  - Will need to convert to attributed strings and changing the font/styling of the author using a range specifier.
  - Will also enable us to single out the bot name and make it clickable (going to that bot)
- Add support for exclusive control with native purchase tools and timer
- Add a "Current Viewers" page to see who is on what robot

Sherlouk, 2017-06-03:
http://vignette1.wikia.nocookie.net/egamia/images/3/32/PlayStation_button_D-pad_Right.svg/revision/latest?cb=20150906185337

For Controls

Sherlouk, 2017-06-05:
Completed:

- [x] Fix issues with camera view not being correct size
- [x] Update the chat area to better use the space and have the main message inline and wrap around the author rather.
  - Will need to convert to attributed strings and changing the font/styling of the author using a range specifier.
- [x] Filter Chat for "This Bot Only" vs "All Bots"
  - This will allow users to quickly toggle between seeing all the chat from everybody, or just people in the same room/robot as you.
- [x] Update the robot chooser to better distinguish between online/offline bots. Potentially defaulting to only show online bots (as you can't do anything with offline ones!)
- [x] Add login capabilities
- [x] Create a splash screen and pre-loader
  - Currently you just get an empty screen whilst the config and socket are loaded and connected. We should make this experience a bit tidier

jillytot, 2017-06-05:
I think the visual feedback for who is pressing what is important. Right now a on the main site you will see dots appear on the buttons as they are pressed. They are tied to specific users and their name colors. 

The buttons have a few states:

This user is pressing the button
Other users are pressing the button:
The robot chooses which direction to go based on current number of inputs per button. 

As far as current viewers, that functionality should be added to the site soon.

Sherlouk, 2017-06-05:
Yeah so I just need to get the visual side of that, all the logic is handled by the server
It's listed green so top priority!

closed 2017-06-07 by Sherlouk