runmyrobot/letsrobot_ios #21: Robot Profile

Sherlouk, 2017-06-07:
- Robot Name
- Creator
- Avatar
- List all robots from the same creator
- Snapshot viewer of that robot
- "Find me on Social Media"?
- Donate button?
- 'Woot' robot?

runmyrobot, 2017-06-09:
Would all of these be on the modal that pops up when you click a robot name?

runmyrobot, 2017-06-09:
Let's leave out the donate button since that can be done with robits/wooting

Sherlouk, 2017-06-09:
@runmyrobot Crossed donate out, fair point.

Not all of this information will be displayed in the modal (not enough space, and want to keep that fairly simple). Though there will be a full profile view which will have everything in it!

Though yes, the idea is that the chat will detect and highlight known robot names which will open a modal with a bit of information and a call-to-action for a larger profile view.

Sherlouk, 2017-06-14:
Work is needed for an easier snapshots API, as well as a user API for information pertinent to the robots owner (e.g. other robots)

Sherlouk, 2017-06-16:
Basic profiles are completed with name, description, last updated and snapshots viewer