runmyrobot/letsrobot_ios #30:  Pay Integration

Sherlouk, 2017-06-08:
Apple Pay is compatible on all phones since/including the iPhone 6.
It's also iPads since the Air 2.

We have a couple of payment points within the application, namely exclusive control (#24), robits (#14) but also could be expanded for things like donations to Lets Robot or the individual caster.

According to [Apples website](https://developer.apple.com/apple-pay/get-started/)
>**Are there additional fees to accept Apple Pay?**
>Apple does not charge users, merchants or developers to use Apple Pay for payments.

The level of integration the Apple Pay API has into the app is very good, it means we don't need to take users away from the app to pay via alternative methods such as PayPal or otherwise. Assuming they've already setup Apple Pay it could be as simple as press "Buy Exclusive Control", authenticate with your touch ID and then control the bot all seamlessly

runmyrobot, 2017-06-08:
Coolio, let me know how the mobile app will need to communicate to the web app that there's been a payment for a service and I can create an endpoint for it

Sherlouk, 2017-06-08:
@runmyrobot So I think for this there are no extra endpoints required (will need to double check that) but the main issues are around simply setting up the Apple Pay with a bank account the money goes into etc.

Will need to talk to you guys about this in more details, especially around the fact certain features ( Pay, Push Notifications, actually releasing etc) can only be done with a paid developer account ($100/yr).

runmyrobot, 2017-06-08:
You'd have to let the web application know that a user bought xcontrol so that the web app can proceed to initiate the actual xcontrol

Sherlouk, 2017-06-08:
@runmyrobot There's already an API for that, checked the other day

runmyrobot, 2017-06-08:
If you mean the POST /internal/braintree/xcontrol/nonce, that endpoint is tightly integrated with braintree's API right now. I think a bunch of changes would have to be made to allow for other kinds of payments such as Apple Pay

Sherlouk, 2017-06-08:
@runmyrobot Ahh right cheers for letting me know - We're quite a whilst from working on this anyway, but closer to the time we can definitely discuss this in more detail.

Updated labels to reflect this!

runmyrobot, 2017-06-08:
awesome possum, yeah, hopefully we'll get to it soon, would be great to have xcontrol working on mobile (as well as robits)

runmyrobot, 2017-06-15:
Payments may not be that hard in fact

Just looked at the braintree API and they may allow accepting payments via PayPal, Apple Pay, Venmo, Credit Card all in one API

Sherlouk, 2017-06-15:
Looks like braintree have an API with their own drop-in UI:
https://developers.braintreepayments.com/start/hello-client/ios/v4

Not sure how we would bridge the gap in between

Sherlouk, 2017-06-16:
https://developers.braintreepayments.com/guides/apple-pay/configuration/ios/v4