runmyrobot/letsrobot_ios #33: Twitch Support

Sherlouk, 2017-06-08:
There is still quite a following on Twitch, is there something here we can do to better the experience and bridge the gap?

A fairly simple solution may be to have a little message in the app which shows when we're live on Twitch, tapping on which would open to the stream?

runmyrobot, 2017-06-15:
I'd prefer if Twitch had a little message that said they're live on Let's Robot :)

Not 100% sure about this one as it'd send users away from our site. I can see a case being made though that it lets users know better who the streamer is and perhaps that'll cause more interest in the Let's Robot view. Not sure

Sherlouk, 2017-06-15:
Dropping out of this release, can talk about it more in the future if someone has strong opinions one way or another.