runmyrobot/letsrobot_ios #35: API: Snapshots for Robot

Sherlouk, 2017-06-09:
Currently the https://runmyrobot.com/api/v1/snapshots is not really feasible to digest within the app.

Ideally we'd be able to specify a robot ID and it'll return an array of snapshots with links to the image/gif (not actually the image stored within the feed in what I can only guess is the bytes representation)

runmyrobot, 2017-06-10:
Could you elaborate why it's not digestible within the app? Is it that the image field in the array of objects returned is a long string of data? Don't remember if that's the case, I would think it should return a URL to the image on S3

Sherlouk, 2017-06-10:
@runmyrobot The API doesn't return a link to the image, it actually returns the image as part of the API - could really do with just a link!

runmyrobot, 2017-06-11:
Yeah, I can try to change it to a link

Sherlouk, 2017-06-14:
Having spoken to Theo today looks like there are two change on the to do list:

https://runmyrobot.com/api/v1/snapshots 
will return links to images rather than returning the images within the feed.

https://runmyrobot.com/api/v1/snapshots/:robotid 
will return the same information but filtered to only show snapshots from the provided robotid

runmyrobot, 2017-06-14:
:thumbsup:

runmyrobot, 2017-06-15:
Going out with the next deploy, you can find the url in both of those endpoints mentioned above.

The second endpoint will return the 10 latest snapshots for the given robot_id.

closed 2017-06-15 by runmyrobot