runmyrobot/letsrobot_ios #38: API: Surface Robot Owner

Sherlouk, 2017-06-11:
Currently on app startup we use the the [`/internal`](https://runmyrobot.com/internal/) endpoint to load such things as the robot array (used for robot list)

This give us a couple bits of information namely the id, name, online status, and sometimes an avatar. (There's also a fixed thumbnail avatar URL which is used as a fallback)

One bit of information that would be nice to show in the app is the robot owner which is currently only possible if we were to do a deep API call for every robot (bit excessive on mobile data!)

Sherlouk, 2017-06-11:
@runmyrobot Would it be possible to get the robot name passed through the internal endpoint as mentioned above?

runmyrobot, 2017-06-11:
This should be relatively simple. I'll try to get to this today

runmyrobot, 2017-06-11:
Looks like robot owner and robot name is already there at the /internal endpoint. They can be found under the "robot" node.

[image]

Sherlouk, 2017-06-11:
@runmyrobot I was referring to the array of robots, not the single robot instance:

[image]

Sherlouk, 2017-06-13:
@runmyrobot This is also more important with the new chat system as this is all based on the owner of the robot. We currently don't know this information without a lot of network requests!

runmyrobot, 2017-06-13:
Added owner to robot list

closed 2017-06-13 by Sherlouk