runmyrobot/letsrobot_ios #40: API: Update Robot Settings

Sherlouk, 2017-06-11:
Can't find the API for this, not sure if it exists

@runmyrobot

runmyrobot, 2017-06-11:
The endpoint is POST /api/v1/accounts/robots which expects an JSON array of robots with each robot's attributes

runmyrobot, 2017-06-11:
So, the request payload would look like this:
```
[
  { 
    camera_id: "90062060"
    charging: false
    createdAt: "2017-04-11T05:22:04.428Z"
    owner: "theodore"
    panels: "[]",
    public: true
    robot_id: "7457915"
    robot_name: "tedTest4"
    status: "offline"
    strong_filtering: true
    subscribers: []
    updatedAt: "2017-04-11T05:22:04.428Z"
  }
]
```

Sherlouk, 2017-06-11:
Thanks for that! Do I need to provide **all** that information in **every** request? Or just the values I update?

runmyrobot, 2017-06-12:
All the information in every request

Sherlouk, 2017-06-12:
Saving this for my sanity (and to remember)

```
{
	"robots": [
	  {
	  	"robot_id": "36061794",
	    "camera_id": "65258911",
	    "charging": false,
	    "createdAt": "2017-06-09T23:27:13.404Z",
	    "owner": "Sherlouk",
	    "panels": "[]",
	    "public": false,
	    "dev_mode": true,
	    "allow_anonymous_control": false,
	    "strong_filtering": true,
	    "robot_name": "SherloukBot2",
	    "status": "offline",
	    "subscribers": [],
	    "updatedAt": "2017-06-12T11:39:59.451Z"
	  }
	]
}
```

Returned

```
{
    "robots": [
        "successfully saved robot"
    ]
}
```

Sherlouk, 2017-06-12:
Implemented Successfully

closed 2017-06-14 by Sherlouk