runmyrobot/letsrobot_ios #42: API: Roles

Sherlouk, 2017-06-11:
In order to properly implement badges in chat we need a way to know the ranks of various users.

https://user-images.githubusercontent.com/15193942/26901323-528d848a-4bcd-11e7-8c71-c11af25007f7.png

Based on the image above, there are 5 badges:
- Staff
- Global Mod
- RoboCaster
- Admin
- Subscriber

We could also probably have
- Chat Mod

Think chat mod, subscriber and robocaster are all in the feeds already in some places (may need to make them easier to find) but others are not surfaced

Sherlouk, 2017-06-14:
@runmyrobot What is the correct way to determine a users role? In order to do moderation tools I need to **at least** understand the difference between at least a user, a robot moderator, and a global moderator.

runmyrobot, 2017-06-14:
If a user is a robot moderator you will see at the GET /internal endpoint, that the node user.robocaster_moderator is true (they should have the Admin badge (or perhaps it would be better changed to a Chat Mod badge))

If a user is a global moderator, you will see that the node user.moderator is true. (they should have the Global Mod badge)

If both of those are false they are a normal user (they should have no badge)

If the node called "is_owner" is true, then they are the robocaster for that page. (they should have the robocaster badge)

If the node at robots[x].subscribed is true, then they should get the Subscriber badge.

If the node called "superuser" is true, then they should get the Staff badge.

Sherlouk, 2017-06-15:
For the sake of moderation tools this is great information, but it won't work for chat badges as I would need to make a request for every user on the site to see what role they are in order to display a badge.

This needs to be sent through via the socket