runmyrobot/letsrobot_ios #43: API: Investigate Relative Time

Sherlouk, 2017-06-12:
The last 30 chat messages which appear in the [internal](https://runmyrobot.com/internal/) endpoint have a relative timestamp of when they were sent.

I need to find out whether this is always "x minutes ago" or whether it changes to a different format once it's longer. (Want to start reading this information into the app, and storing it for the new messages, in order to display a richer chat experience)

Sherlouk, 2017-06-12:
@runmyrobot Maybe you could answer this?

I've just noticed it does "x seconds ago" but it does that at 79 seconds. Where is the boundary for that? 111 seconds ago -> 2 minutes ago?

Is there any boundary above minutes ago? Hours ago? Days ago? etc?

Sherlouk, 2017-06-14:
Low priority, skip for now.

runmyrobot, 2017-06-15:
```
function timeSince(date) {

  var seconds = Math.floor((new Date() - date) / 1000);

  var interval = Math.floor(seconds / 31536000);

  if (interval > 1) {
    return interval + " years";
  }
  interval = Math.floor(seconds / 2592000);
  if (interval > 1) {
    return interval + " months";
  }
  interval = Math.floor(seconds / 86400);
  if (interval > 1) {
    return interval + " days";
  }
  interval = Math.floor(seconds / 3600);
  if (interval > 1) {
    return interval + " hours";
  }
  interval = Math.floor(seconds / 60);
  if (interval > 1) {
    return interval + " minutes";
  }
  return Math.floor(seconds) + " seconds";
}
```

Sherlouk, 2017-06-15:
@runmyrobot How difficult would it be to return the absolute time rather than relative time? Makes my life a lot easier than having to account for all these situations (and is just more powerful honestly)

runmyrobot, 2017-06-15:
Not very difficult, I'll put it on my todo list