runmyrobot/letsrobot_ios #45: API: Preference Defaults

Sherlouk, 2017-06-12:
Currently the API seems to only show certain values if they're not the default. For example a description field only shows if it's not empty, and the "dev_mode" preference only seems to show if it's turned on.

@runmyrobot 

Can you confirm to me what the defaults are for:
"public", "mute", "strong_filtering", "allow_anonymous_control", and "dev_mode" so I can ensure the robot settings page in the app reflect this

runmyrobot, 2017-06-14:
default values:
public: true,
mute: false,
strong_filtering: false,
allow_anonymous_control: true,
dev_mode: false

runmyrobot, 2017-06-14:
I'll also add to the todo list that we can have the API return a fixed schema regardless of null/default values

Sherlouk, 2017-06-14:
I'll update the code tomorrow to ensure it reflects this, it'll be a seamless transition when the schema updates!

closed 2017-06-15 by Sherlouk