runmyrobot/letsrobot_ios #50: Update subscriptions to be a two way relationship

Sherlouk, 2017-06-13:
A robot needs an array of subscribers (Array of users)
A user needs an array of subscriptions (Robots, or a simpler struct)

Using "subscription_state_change" socket event, these lists should be both be updated. They should update one another and always be synced.

Also need to update the subscription count label at UI label if this value ever changes.

runmyrobot, 2017-06-14:
Currently there only exists the "A robot needs an array of subscribers (Array of users)". I gather a user's subscriptions by querying for robots that have that user as a subscriber, this way I don't have to worry about keeping things in sync (although it taxes the DB I suppose). 

I can add the list of a robot's subscribers to /api/v1/robots/:robot_id if you'd like. Later, we can move to a new endpoint for that (meant for authorized clients only (since it's not completely necessary for everyone to see everyone's subscribers))

Sherlouk, 2017-06-14:
So previously this couldn't be done due to separation I had between public and private robots being in silos unaware of one another.

In theory though that may now be possible as I've changed the way this all works today. Will continue to investigate

Sherlouk, 2017-06-15:
Fixed

closed 2017-06-15 by Sherlouk