runmyrobot/letsrobot_ios #51: API: User Profile

Sherlouk, 2017-06-14:
there appears to be an endpoint at `/api/v1/accounts/:name` but only ever returns the name from what I can see?

Could do with an API which returns the users name, description, maybe list of their subscriptions, an array of their robots, etc

runmyrobot, 2017-06-14:
I've put this on my todo list

runmyrobot, 2017-06-15:
Going out with the next deploy will be more information about the account exposed via that API above.

The schema would look like this:
[image]

Sherlouk, 2017-06-15:
@runmyrobot So the only thing I can think of at the moment is an array of their robots. This is not a massive necessity though since owner has now been added to `GET /internal` so can just filter that!

Will work on this

Sherlouk, 2017-06-15:
Work has been completed (I think) I'll leave issue open until next deploy where I can validate functionality

Sherlouk, 2017-06-15:
Whilst doesn't fix the entire issue of roles (#42) having "superuser" in the public user API would be nice also

Sherlouk, 2017-06-16:
Think for now I'm going to skip full user profiles and just block out the button for going to to it (keeping the modal one).

There just isn't enough information to show to fill out a page. The big deal is a robots list, but if you've got no robots [which is most people] then all I can show in your user description. Which again, most people don't even set!

Will let someone else decide this